lexus2k/tinyproto #3: empty payload frame cannot be parsed as a Tiny::Packet

alieslam, 2018-12-30:
When I tried to parse a frame with no payload just the UID using the  regular buffer (technique no.1) it seems working normally. While when I tried using Tiny's Packet class (technique no.2) it didn't seems to work, and it just ignores the whole frame. Am I missing any information? 

Hint: I am using the following functions in both techniques to initialize hdlc instance
```
hdlc.beginToSerial(); /* using Arduino's Serial Instance to parse and send data */
hdlc.enableCrc16(); /* Enabling CRC16 to frames */
hdlc.enableUid(); / * Enabling Uid for Tiny::Packet */
```

Technique no. 1
```
char buff[100]={0};
Tiny::Proto hdlc;
int len = 0;
len = hdlc.read(buff,sizeof(buff),TINY_FLAG_NO_WAIT);
if (len > 0)
{
    /* Process the Data */
}
```

Technique no. 2
```
char buff[100]={0};
Tiny::Proto hdlc;
Tiny::Packet myPacket(buff,sizeof(buff));
int len = 0;
len = hdlc.read(myPacket, TINY_FLAG_NO_WAIT);
if (len > 0)
{
    /* Process the Data */
}
```

alieslam, 2018-12-31:
@lexus2k Should the fix also be supported with ProtoLight read packet function?
https://github.com/lexus2k/tinyproto/blob/master/src/TinyLightProtocol.cpp#L65

lexus2k, 2019-01-01:
@alieslam I believe no, since ProtoLight version doesn't support UIDs. As for zero-length packets ProtoLight version tiny_light_send() and tiny_light_read() always run in blocking mode and return negaive value if they failed, in all other cases they return 0 or N depending on length data being sent or received. So, they should be OK.