lexus2k/tinyproto #6: Packaging of source code for Arduino does not work

nmaas87, 2019-07-19:
I wanted to try your 0.7_dev_rework - and used the ``gen_arduino_release.bat`` to package the current development code, however, this only gave a lot of file not found errors.

Also for the master branch, the https://github.com/lexus2k/tinyproto/tree/0.7_dev_rework#setting-up-for-arduino description does not fit anymore.

Could you have a look into it? Thanks a lot for your awesome work, I think it will fit my usecase 👍

lexus2k, 2019-07-19:
Hi, Nico.
I'm going to commit fix today. Right now I have issues with ttyUSB0 device and trying to fix them.

lexus2k, 2019-07-20:
You can try once again now, I pushed all local changes to github

nmaas87, 2019-07-20:
Hi thanks for your awesome work,

sadly the bat file to create the package does not work as it does not reflect the current state of the library. However, I found out that @santicastro did fix quite a bit on his fork, especially the 3 commits from 05.02.2019 and 03.05.2019 seem to be interesting: https://github.com/santicastro/tinyproto/commits/master

The f7b4c71c7217369bba8b143714a485d28e4a4461 bascially generates the bat file for linux systems. I reworked it a bit and did also reduplicate it for windows, except for the zip packaging. I did also update both library files and moved the example folder within the packing so that it comes up without another "src" as upper directory within the Arduino IDE under the examples section of tinyproto.

I should rewrite the howto use to explain better to new users on how to use your lib :)

I send you a pull request for the time being :)

nmaas87, 2019-07-20:
OK and I updated the README.md as well now for the PR :)

lexus2k, 2019-07-21:
Thank you for your help, I merged PR.
Also, please note, that with current library structure it is possible to use it for Arduino as is.

nmaas87, 2019-07-21:
Perfect. I opened up another PR, I wanted to merge the Fix for Arduinos with multiple Hardware Serial Interfaces as well, which has also been done by @Santiago. I also found an return true for an void function to disableCrc, but I saw that you already found this yourself yesterday 👍

closed 2019-07-22 by nmaas87