lexus2k/tinyproto #10: Null dereferencing, examples to small and not working.

Zataras, 2020-02-06:
Hi, 

At `hdlc_run_rx_until_read()` there is no check for `( readcb != nullptr )` before calling it, so program is unexpectedly crashing when it's not initialized. It could be at least checked with assertion to help debugging.
And examples are too short, I think it's required to call at least `proto.begin()` to begin using proto( to avoid null dereferencing ), so why it's not mentioned in examples? They would be more helpful then.

Anyway, I'm starting to work with your library and it looks like good job.

Regards,
Zataras

lexus2k, 2020-02-07:
Hello, thank you for the notes. `beginToSerial`, which is used in examples, does the same work as `begin` method. You can find implementation in header files.
As for `hdlc_run_rx_until_read`, I can add to documentation that readcb cannot be NULL, but if you really need it to be NULL, it is ok to add verifications.
Let me know, what is the best way to your mind.

Zataras, 2020-02-18:
Hi, thanks for quick response. I'm trying to use tinyproto on Linux with usb<->uart converter. I think that predefined functions like beginToSerial are valid not for all platforms, is that right? They are also not mentioned at "Easy to use" paragraph. 
With my idea of assertions - I recall this, as assertions are not platform-flexible I guess. But error code would be easy to return on nullptr.

lexus2k, 2020-02-18:
Hello,

Well, could you please clarify your setup? As far as understand, your setup is:

PC<->USBPort<->USB2TTLconverter<->UART<->Your device.

So, what is your device?
I found that some USB2TTL converters are not good as they should be, even if they are based on FTDI232 chip.
Next question, do you have some logic analyzer?

`beginToSerial()` should work for most of Arduino development boards.

closed 2020-03-11 by lexus2k