lexus2k/tinyproto #13: What flags needed in build when using tinyproto as library

Zataras, 2020-09-10:
Hi, 

I could not figure out why tinyproto was behaving much differently in my application that in the loopback example and I came to conclusion that this is about my project build flags. How should I build my project to have tinyproto working correctly inside? Currently when I run tinyproto functions I get e.g. memory related exceptions, or just application crashes. It does not happen why I do not use tinyproto, also it does not happen in loopback example build with your makefiles. I copied loopback code directly to mine project and it's crashing not specifying the reason.

Regards

lexus2k, 2020-09-10:
Hello, could you please provide the example application and describe your platform?

Zataras, 2020-09-11:
Hi, I've created example project in CMake. I guess there is something about build differences when I use `make` for tinyproto and another config from CMake for project using the build tinyproto library. This is the project https://github.com/Zataras/ExampleWithTinyproto. Currently I'm getting Segmentation fault when I run tiny_loopback example built externally with CMake. Please help, maybe you'd have some idea what I'm missing here.

lexus2k, 2020-09-12:
Unfortunately, I failed to build using my Cmake 3.10, since it doesn't have `target_link_directories` command.
I will try other ways to build.
Meanwhile, could you please try latest release of the library? It supports cmake out of the box and has some fixes.
Did you try to run tiny_loopback under gdb?

lexus2k, 2020-09-14:
I've fixed compilation in your CMake file:
```.patch
git diff CMakeLists.txt
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 8c0d70f..b8e561a 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -10,23 +10,30 @@ project(ExampleWithTinyproto VERSION 0.0

 set(BASE_DIR ${CMAKE_SOURCE_DIR})

+find_library(TINYPROTO_LIBRARY
+    NAMES
+        tinyprotocol
+    PATHS
+        ${BASE_DIR}/lib/tinyproto/bld
+    REQUIRED)
+
 file(
-       GLOB_RECURSE
+       GLOB_RECURSE
        SOURCES
        src/*.*pp
        ${BASE_DIR}/lib/tinyproto/tools/serial/serial_linux.c #OpenSerial() etc.
-)
-
-add_executable(
+)
+
+add_executable(
        ${PROJECT_NAME}
-       ${SOURCES}
+       ${SOURCES}
 )

 target_include_directories(${PROJECT_NAME} PUBLIC ${BASE_DIR})

 #TinyProto requires:
 target_include_directories(${PROJECT_NAME} PRIVATE ${BASE_DIR}/lib/tinyproto/src)
-target_link_directories( ${PROJECT_NAME} PUBLIC ${BASE_DIR}/lib/tinyproto/bld )
-target_link_libraries( ${PROJECT_NAME} PUBLIC libtinyprotocol.a pthread )
+# target_link_directories( ${PROJECT_NAME} PUBLIC ${BASE_DIR}/lib/tinyproto/bld )
+target_link_libraries( ${PROJECT_NAME} PUBLIC ${TINYPROTO_LIBRARY} pthread )

 target_compile_features(${PROJECT_NAME} PUBLIC cxx_std_11)
```

And yes, if I compile the project with the libtinyprotocol.a binary provided by you, I've got crash. But if I compile tinyprotocol library via `make clean && make` and then recompile your project, nothing gets crashed.
So, please check building process on your side, and start to use latest version of the library

Zataras, 2020-09-14:
Thank you very much for your response. I'm sorry, I have no idea why I did not check if there are updates on tinyproto project. Thank you for your fix. I will also use latest library source code.

Zataras, 2020-09-14:
It's working just with new version - 0.10.1, issue resolved. Thanks again!

closed 2020-09-14 by Zataras