lexus2k/tinyproto #18: Examples vs Tinyproto documentation

timov50, 2021-05-03:
Hi, 

I'm a bit noob with C++ and have some difficulties to understand how to use Tinyproto. 

In the Tinyproto documentation there is mentioned that higher level protocols needs 4 user defined callbacks.

- `int write_func_cb(void *user_data, const void *data, int len);`
- `int read_func_cb(void *user_data, void *data, int len);`
- `int on_frame_read(void *user_data, void *data, int len);`
- `int on_frame_sent(void *user_data, const void *data, int len);`

Is the documentation (in https://codedocs.xyz/lexus2k/tinyproto/index.html) out of sync with the main branch because in the examples I find references to the `on_frame_read` and `on_frame_sent`functions, but not to the `write_func_cb` or `read_func_cb`functions? 

Examples where `on_frame_read` or `on_frame_sent` are used:
`./examples/linux/hdlc_demo_multithread/hdlc_demo_multithread.cpp`
`./examples/esp32_idf/spi/master/main/app_main.c`
`./examples/esp32_idf/spi/slave/main/app_main.c`

Or have I missed something essential?

-timo-

lexus2k, 2021-05-03:
Hi,

Yes, you're right. Some pages in the documentation are outdated. The source code for the page, you've pointed out is here: https://github.com/lexus2k/tinyproto/blob/master/src/mainpage.dox. Feel free to update it, and create pull request, or I will fix it later.
The most correct "documentation" is examples folder. It is always up to date.

Thank you

timov50, 2021-05-05:
Thanks for quick response. 
I’m afraid I don’t understand the design of tinyproto well enough to touch the documentation.

E.g., what is going here (esp uart loopback example)?
`proto.run_tx([](void *p, const void *b, int s) -> int { return uart_write_bytes(UART_NUM_1, (const char *)b, s); });`

I do not understand role ole `p*` as the `uart_write_bytes` function (esp-idf) does not use it. 

My understanding is that the first part of the lambda is coming from here:
`    typedef int (*write_block_cb_t)(void *pdata, const void *buffer, int size);`
The `*pdata` is definition:
`@param pdata - pointer to user private data - absent in Arduino version`

Q: What is the purpose of the `*pdata` argument? Legacy?

lexus2k, 2021-05-06:
Hi @timov50 

`*p` is just a pointer to the user data, which can be specified via `setUserData()`, if you need it. So, in the example `*p` is not used.