lexus2k/tinyproto #20: Software serial support

githubforsank, 2021-07-26:
Hi @lexus2k & contributors,
I have a situation where I want to reliably send high-speed data from RPi and Arduino. 
The problem is I am using an Arduino Nano which has one HW Serial. 
The problem is I cannot connect the RPi to the HW Serial because it creates issues in uploading
sketches with the Serial USB, so I am using Software Serial to communicate with the Pi. 
Now exploring very few solutions I stumped across this beautiful library as it will serve the purpose
of the Pi communicating with TinyProto in python as well. 
But I see now the library uses only Hardware Serial library and provides hardware redirects
`beginToSerialX() wehre X-> No of serial ports. `
Is there any issue with adding something like 
`beginToSoftwareSerial(Rx, Tx).` and begin with SoftwareSerial.
Please let me know if is possible through some code hack in the stable version. 
I would really like to see this as an official feature. 
Let me know what you think about it. 

Thank you,
Sasank Panda

lexus2k, 2021-07-26:
Hi Sasank,

The library is flexible enough to let you use SoftwareSerial without any changes to the library itself.
See how `beginToSerialX` method is implemented inside the library:

```.cpp
    inline void beginToSerial1()
    {
        begin([](void *p, const void *b, int s) -> int { return Serial1.write((const uint8_t *)b, s); },
              [](void *p, void *b, int s) -> int { return Serial1.readBytes((uint8_t *)b, s); });
    }
```

Thus, instead of calling `beginToSerialX` method in your sketch, use can call just `begin` method with your own implementation of callbacks:

```.cpp
    #include <SoftwareSerial.h>
    SoftwareSerial mySerial(2, 3);
    ...
    proto.begin([](void *p, const void *b, int s) -> int { return mySerial.write((const uint8_t *)b, s); },
              [](void *p, void *b, int s) -> int { return mySerial.readBytes((uint8_t *)b, s); });
```

Let me know, if you have any issues with the example above.
Also, I would recommend to carefully test Software Serial library communication, because unlike Hardware implementation, the Software one may miss some bits due to higher load off AVR MCU.

Best regards,
Alexey

githubforsank, 2021-07-27:
Thank you @lexus2k ,
I was unsure of the implementation but now I have a clear idea.
I will try to implement this and let you know of any issues. 
Thanks for the prompt response and direction.
Thank you,
Sasank Panda

closed 2021-07-29 by githubforsank