lexus2k/tinyproto #23: Unreliable communication with tinyproto

githubforsank, 2021-07-29:
I am unable to get reliable data transmission
at a bit higher baud rate: 115200, 57600.
Enabled crccheck16() on both sides. 

Send :
Rpi python
```
ser = serial.Serial('/dev/ttyS0', 115200, parity = serial.PARITY_NONE, timeout= 0)

while True:
    buf = bytearray([ 0x7E, 0xFF, 0x3F, 0xF3, 0x39, 0x40, 0x7E  ])
    buf = "SAM\0"
    buf = "HELLO-WORLD\0"
    print("Writing byte array to serial")
    print(p.put(buf)) #bytearray([ 0x7E, 0xFF, 0x3F, 0xF3, 0x39, 0x7E  ])
    # print(p.tx())
    ser.write(p.tx())
```

Receive: 
Arduino
```
#include <TinyProtocol.h>
#include <SoftwareSerial.h>

int buff[256];
tinyproto::Hdlc proto_hdlc(buff, 256);
SoftwareSerial mySerial(A3, A2);

void hdlcRecieveCallback(tinyproto::IPacket &pkt)
{
    //Recive and process packets
    Serial.println("HDLC Recieve callback");
    Serial.println(pkt.getString());
    Serial.println("Recieved data size : " + String(pkt.size()));
    Serial.println("Max buffer size : " + String(pkt.maxSize()));
    Serial.println("=========================");
}
void setup() {
    /* No timeout, since we want non-blocking UART operations. */
    Serial.setTimeout(0);
    mySerial.setTimeout(0);
    /* Initialize serial protocol for test purposes */
    Serial.begin(115200);
    mySerial.begin(9600);
    proto_hdlc.setReceiveCallback(hdlcRecieveCallback);
    proto_hdlc.begin();
    proto_hdlc.enableCrc16();
}

void loop()
{

    size_t len =   mySerial.available();  
    // char bytes[128];
    if (len) {
        uint8_t bytes = mySerial.read();
        proto_hdlc.run_rx( &bytes, 1 );
        // mySerial.readBytes(bytes,len > 128 ? 128 : len);
        // proto_hdlc.run_rx( bytes, len > 128 ? 128 : len ); // run FD protocol parser to process data received from the channel
    }
}
```

Baud: 115200
Zero reliability

[image]

More reliable than above.
But still unacceptable to the requirement.
Baud : 57600
[image]

Most reliable
Still too many error packets
Baud : 9600
[image]

Am I doing something wrong. 
or the observation is correct.

Thank you in advance.
Sasank Panda

lexus2k, 2021-07-30:
Hi Sasank,

did you try examples from `examples` folder? Do they work on your setup?
As I wrote earlier in another github issue #20: _Also, I would recommend to carefully test Software Serial library communication, because unlike Hardware implementation, the Software one may miss some bits due to higher load off AVR MCU._
Even printing debug information to hardware Serial `Serial.println("HDLC Recieve callback");` in the `hdlcRecieveCallback` callback eats the resources of AVR MCU.

Since, you're using Software Serial on the weak micro controller, this could be a root cause. You can try to switch to more powerful hardware.

githubforsank, 2021-08-02:
Yes, I tried, 
I did finally change to AltSoftSerial the second-best to hardware as I managed
to get pin 8,9 free. There I observed I was writing to the buffer too fast (probably faster CPU of RPi ), than the Arduino can empty the buffer. I managed to get good
reliability adding some delay although I am not convinced with the solution. 
But yes no issues with the library functionality. 

Thank you , 
Sasank Panda

closed 2021-08-02 by githubforsank