lexus2k/tinyproto #25: License?

gschmottlach-xse, 2021-11-12:
I stumbled upon your project and was intrigued. I have both hobby and work projects where I need a simple, robust protocol, to communicate messages over a serial/SPI interface. It appears you put a lot of effort into tinyproto and I suspect it would be useful for many of my projects. My enthusiasm, however, significantly diminished when I noticed the license: LGPLv3. Although fine for home/hobby projects, it's not something I could integrate in work/commercial projects due to the copy-left requirements. For deeply embedded projects it's impossible to separate tinyproto code from a proprietary application. For this reason MIT/BSD/Apache licenses are more desirable and easier to work with.

If you don't have strong feelings about LGPLv3 and copy-left issues, I wonder if you would consider changing the license to one that is more permissible? I tend to avoid expending time and energy learning and integrating a library if I have to worry about where I can eventually utilize it. With that said, since you wrote the code, you get to chose where/how people use it. I just wanted to make you aware, in case you weren't, how your license choice impacts the adoption by a broader audience of developers. Perhaps this is as intended . . . but I hope you might reconsider?

Regardles, thanks for making it available!

lexus2k, 2021-11-18:
Hello, Glenn.

The LGPL license was chosen intentionally. In many of my open-source projects I use the MIT license, and I'm OK with it. From one side the MIT license doesn't require disclosure of the source code, where a library is integrated to. From the other side, the practice gives the understanding that almost no changes and improvement are contributed to the library from side developers.
Yes, you're right, the tinyproto library took much effort to make it work, and I have more ideas to improve it, but I almost do not have free time. Thus, I would like to see how it is used in the projects, and the LGPL license should force other developers to publish their code (if they follow the terms of the license).
The tinyproto project officially has only LGPL license while it is still possible to have a separate Software License Agreement between the commercial company and me.
As an example of this approach, QT project has dual-license for both open source and commercial use.
If it is not a secret, which company do you work for?

Best regards

gschmottlach-xse, 2021-11-18:
Hi Aleksei -

I do understand your motivations and respect them, but in practice, from my experience the LGPLv3 turns out to be an unnecessary burden for use in proprietary applications. I've worked for very small startups to (currently) a very large technology company. My current employer makes every effort to follow the terms of the license but to get something like a LGPLv3 component integrated into a bare-metal application stack (where it can't simply be linked as a shared library) is a non-starter. First, getting legal involved to review and understand this license in the environment in which your library might be used is a major headache (and prohibitively time-consuming). Second, unfortunately, the value that this component might offer if licensed separately (e.g. a paid license) is not significant enough to justify pursuing (again, through legal and purchasing . . . a long drawn out ordeal). If this library were an integral component for a proprietary application being develop (e.g. an entire HMI framework like QT) that was going to serve as the foundation for a software suite going forward, a separate (paid) license might be justified since you'd also get a fair amount of support with it. As you said, like many of us, free time to develop these small (yet useful) libraries is very difficult to find. It's an expression of passion for development. Paying for a license implies providing support and it's not clear you have the time/energy to fullfil that part of the obligation (which I totally understand).

Honestly, I'm not looking to develop this library further. With that said, when I run across bugs or make minor improvements I usually offer them back to the original developer as a courtesy (and thanks) for their efforts to make their software available for others to use. Plus, it's only to my benefit if my fixes or small improvements get re-introduced into the mainline so I don't have to maintain a separate private fork. For me, personally, I just get a kick out of hearing how others use the stuff I develop on the side. If I were to become a free-lancer and had to pay my own bills, then sure, I'd license my software in some manner to materially benefit from my efforts (perhaps through support contracts etc...). Perhaps we don't agree on this topic and you're entirely within your right to license your creations however you like - it's your choice. For me, however, LGPLv3 introduces enough of a legal burden and uncertainty (in the context of a proprietary application) that I naturally shy away from it. Since the bulk of the software I develop is proprietary in nature (since someone pays me to write it), I have fewer opportunities where I could use your component except for hobby projects done outside of work. Since I cannot freely use this component both inside/outside of work, the effort it takes to learn and integrate your work isn't as broadly applicable. As a consequence, I'd more likely roll my own solution or find a package with a less restrictive license that I extend myself.

Anyway, I appreciate your candor. I hope you continue your developments . . . it looks like a nice package.

Best regards . . .

closed 2021-11-23 by lexus2k