lexus2k/tinyproto #27: Pi Pico serial1 problem

asevim, 2021-11-24:
Hi @lexus2k & contributors,
Firstly thank you this qualified work. I would like to use this protocol with Pi Pico. But I have some issues about Serial1 When used on Pi Pico. My hardware setup is like this with an Arduino Nano board.
[image] 
I ran loopback code with hd parameter on OrangePi SBC, RX and TX leds blinked both modules(USB to RS485 and TTL to RS485). It works without problem. But only When I changed Pi Pico instead of Nano. It isn't work. TX led didn't blink although RX led blinked, on TTL to RS485. But When I try Serial1.print("Hello") TX led is open. My Pi Pico code is like this.

```ruby
#include <Arduino.h>
#include <TinyProtocol.h>

void onFrameIn(uint8_t *buf, int len);

uint8_t buffer[64];

Tiny::ProtoHd  proto(buffer, sizeof(buffer), onFrameIn);

void onFrameIn(uint8_t *buf, int len)
{
    uint8_t tx_buffer[64];
    memcpy( tx_buffer, buf, len );
    proto.write( (char *)tx_buffer, len );
}

void setup() {
    Serial1.setTimeout(10);
    Serial1.begin(115200);
    proto.enableCheckSum();
    proto.beginToSerial();
}

void loop()
{
  proto.run();
  //Serial1.print("Hello"); //Open TX led
}
```

 Also I tried USB communication between Pi Pico and Orange Pi with Tiny FD. It works. I defined Serial.begin(115200) for USB communication on Pi Pico.

Arduino Core basic documentation for Nano RP2040 same as Pi Pico:
https://docs.arduino.cc/tutorials/nano-rp2040-connect/rp2040-01-technical-reference

Best regards,
Ahmet

lexus2k, 2021-11-25:
Can you switch to latest version of the protocol on the master branch.
Half-duplex is removed from the library. There are only 2 options now: plain HDLC framing with CRC checks and Full-Duplex protocol.

asevim, 2021-11-25:
Normally I worked on version 0.9.3 in PlatformIO. I read your response then I uploaded 0.12.1. I tried many kinds of tests. Problem is the same, It doesn't work both light and FD with Pi pico Serial1. It worked with Arduino Nano but didn't work with Pi Pico. For example sketch_light.ino code tried in both, Nano board blinked TX led but Pi pico TX led didn't blink.

lexus2k, 2021-11-25:
Ok, I understand. Then I have the question: Which package do you use in Arduino IDE for Pi PICO? Arduino Mbed OS RP2040 Boards?

asevim, 2021-11-26:
PlatformIO uses official Arduino Mbed Core framework. I can call Mbed OS API's in code. Some information in [Arduino blog](https://blog.arduino.cc/2021/04/27/arduino-mbed-core-for-rp2040-boards/). Someone use another Arduino Core for RP2040 like [this](https://github.com/earlephilhower/arduino-pico).

lexus2k, 2021-11-28:
Hi,

it's me again. I don't have the PICO board by hand, but if you look to the beginToSerial() method implementation, you will find:

```.cpp
void Light::beginToSerial()
{
    Serial.setTimeout(100);
    begin(writeToSerial, readFromSerial);
}
```

So `beginToSerial()` uses `Serial` port, but NOT `Serial1`. These ports are different. So, you need to use the method, corresponding to your port, I believe it is `beginToSerial1`.

asevim, 2021-11-29:
Hi,
Sorry your waste of time. It solved after that defined `#define HAVE_HWSERIAL 1`. I'm novice in programming. Thank you for all help, i solved through your help. But I have a question. Why did you delete Half duplex in TinyProto. It works in my tests on RS485. Is there any plan in future again add Hd.

lexus2k, 2021-11-29:
All higher level protocols: FD, Light, HD, - are built on top of `src/proto/hdlc`. In the latest versions on the master and cpp_api branches low level hdlc has been reworked greatly. So, I just didn't have enough time to rework and support the half-duplex version.
So the answer is simple: just lack of time.

asevim, 2021-12-01:
Okay, thank you for reply and interest. I hope you rework on HD in the long term.
Best Regards.

lexus2k, 2021-12-01:
I'm glad that finally you solved the problem with Pi Pico board.

> I hope you rework on HD in the long term.

Can you submit new issue on that, and give there a little more points, why HD is more preferably than FD or plain HDLC frames in your case?
Thank you.

asevim, 2021-12-02:
I would like fast installation on RS485. In the future I may work on HDLC frame.
So the answer is simple: I am beginner at now :)

closed 2022-01-06 by asevim