lexus2k/tinyproto #28: Questions on connection management

chenlijun99, 2021-12-18:
Hi @lexus2k & contributors.
First, thank you for this excellent library!

I'm trying to implement a socket-like wrapper of tinyproto FD for my project, both in C/C++ and in Python. What I'm having a hard time doing is the connection establishment. In particular I could not find these things:

* on_connect/on_disconnect callback. I'm aware that there is the `tiny_fd_get_status` function, but I would like to avoid polling.
* Even if I used `tiny_fd_get_status` AFAIK this function is missing in the Python wrapper.

Have these things ever been considered? Or are there valid reasons not to have them? Would you accept a PR?

lexus2k, 2021-12-19:
Hi Lijun

You're right, there is no way to use handlers for connect/disconnect events. `tiny_fd_get_status()` is the only way now to get the link status. But I think, it is good idea to add on_connect/on_disconnect callbacks to the tinyproto library.

Python PR is approved.

Best regards

lexus2k, 2022-01-10:
Hi,

I added connect/disconnect callback. So, you can use them in your code now.
Just pull the latest changes from the master branch.

closed 2022-01-10 by lexus2k

chenlijun99, 2022-01-11:
Thank you for your work!