lexus2k/tinyproto #35: Build with visual studio works, but unable to do a proto begin --> m_handle=null

Richju07, 2022-04-27:
Hi Team,

I have an issue when I m using the example project tiny_loopback build with Microsoft Visual Studio. Every time that I tried to start proto.begin(), it 's not possible to create the m_handle. It seems that program failed at :

/* Next we allocate some space for peer-related data */
ptr = TINY_ALIGN_BUFFER(ptr);
protocol->peers_count = peers_count;
protocol->peers = (tiny_fd_peer_info_t *)ptr;
protocol->next_peer = 0;
ptr += sizeof(tiny_fd_peer_info_t) * peers_count;

if ( ptr > (uint8_t *)init->buffer + init->buffer_size )
{
LOG(TINY_LOG_CRIT, "Out of provided memory: provided %i bytes, used %i bytes\n", init->buffer_size,
(int)(ptr - (uint8_t *)init->buffer));
return TINY_ERR_INVALID_DATA;
}

In tiny_fd.

The strange things, it's I have not the same behavior if I'm using clang ++ and ninja to build the project.

Is someone found this issue too?

Regards,
Julien R .

lexus2k, 2022-05-02:
Hi Julien,

Could you please provide complete example to reproduce the issue?
Is your platform Windows 64-bit based on x86 architecture?

Richju07, 2022-05-04:
Hi Lexus2k,

I'm using visual studio 2017 to build in win 32. I'm using the version of the master branch.

Regards,
Julien R.

lexus2k, 2022-05-05:
I compiled the project in Visual Studio from master branch:

[image]

As you can see, the handle is not null. 
Can you provide more details? For example, describe the parameters you use to launch tiny_loopback?