lexus2k/tinyproto #40: Connection breaks down if nothing is sent for >= keep alive timeout

chenlijun99, 2022-07-10:
I would expect Tinyproto to schedule some S frames as keep alive frames.

lexus2k, 2022-10-27:
Tinyproto already has this:

https://github.com/lexus2k/tinyproto/blob/164eecebcda1734469456f98c910494534e53483/src/proto/fd/tiny_fd.c#L999

If you consider a bug, let me know

chenlijun99, 2023-03-24:
Hi. Sorry for the lack of feedback.

Indeed, there is already a mechanism to send keep alive frames. I can see that sometimes it works, sometimes it doesn't. If I understand the code correctly, once a keep alive frame has been sent, if `tiny_fd_connected_check_idle_timeout` is immediately re-called (because `tiny_fd_get_tx_data` is immediately re-called), before the response to the keep alive frame is received, then we immediately enter this branch

https://github.com/lexus2k/tinyproto/blob/164eecebcda1734469456f98c910494534e53483/src/proto/fd/tiny_fd.c#L1001-L1005

and disconnection is asserted.

Maybe a sort of keepalive response timeout should be considered?