lexus2k/tinyproto #43: riscv processor based esp32 series support

Tizio1234, 2024-02-07:
When trying to build for esp32c3(or any riscv processor based esp32) tiny_types.c and tiny_types.h don't include the esp32/esp32_hal.inl(or .h, I don't remember).
This can be fixed by adding to the line "#elif defined(__XTENSA__)" " || defined(__riscv)".

Should I create a new branch with this simple fix and do a pull request?

lexus2k, 2024-02-10:
@Tizio1234 

The compilation for ESP32 C3 board is successful.
[image]

Are you using some specific board support package?

Tizio1234, 2024-02-10:
I'm not using the Arduino IDE, I'm using esp idf 5.2 directly, other than this, I get a watchdog timer timeout when calling proto.run_tx... for the second time.
Can the idf version be the cause of this?

lexus2k, 2024-02-11:
@Tizio1234 This could be. The library was tested with idf 3.X and 4.X, but not with the 5.X.
Do you have code example, which causes the problem?

Tizio1234, 2024-02-11:
Here's main.cpp:
```
#include <stdio.h>
#include <string.h>
#include <unistd.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"

#include "driver/uart.h"
#include "driver/gpio.h"
#include "esp_log.h"

#include <TinyProtocol.h>

static const char* TAG = "prova tinyproto";

static const char message[] = "hello from tinyproto hdlc";

#define TINY_MULTITHREAD
#define WINDOW_SIZE (7)
#define GENERATED_PACKET_SIZE (64)
#define BUF_SIZE (512)

#define TP_UART_NUM UART_NUM_1
#define TP_UART_BAUDRATE 115200
#define TP_UART_TX GPIO_NUM_27
#define TP_UART_RX GPIO_NUM_26
#define TP_UART_BUF_SIZE 1024

#define TP_TX_BUFF_SIZE 256

tinyproto::FdD proto(tiny_fd_buffer_size_by_mtu(256, WINDOW_SIZE));

static QueueHandle_t uart_queue;

static TaskHandle_t uart_event_task_handle = NULL;
static TaskHandle_t uart_tx_task_handle = NULL;

static void uart_event_task(void *pvParameters){
    uart_event_t event;
    for (;;)
    {
        if (xQueueReceive(uart_queue, &event, portMAX_DELAY))
        {
            if (event.type == UART_DATA)
            {
                char* data = (char*)malloc(event.size);

                uart_read_bytes(TP_UART_NUM, data, event.size, portMAX_DELAY);
                
                int remaining = (int)event.size;
                int read = 0;
                while (remaining > 0)
                {
                    int tmp = proto.run_rx((void*)&(data[read]), remaining);
                    remaining -= tmp;
                    read += tmp;
                }

                free(data);
            }
        }
    }
}

static void uart_tx_task(void *pvParameters){
    void *buff = malloc(TP_TX_BUFF_SIZE);
    if (buff == NULL)
[... 142 lines cut ...]
0x400d71e5: hdlc_ll_run_rx at /home/tommaso/esp/workspaces/tinyproto_ws/tinyproto_52/components/tinyproto/src/proto/hdlc/low_level/hdlc.c:516
0x400d6865: tiny_fd_on_rx_data at /home/tommaso/esp/workspaces/tinyproto_ws/tinyproto_52/components/tinyproto/src/proto/fd/tiny_fd.c:841
0x400d5e65: tinyproto::IFd::run_rx(void const*, int) at /home/tommaso/esp/workspaces/tinyproto_ws/tinyproto_52/components/tinyproto/src/TinyProtocolFd.cpp:97
0x400d5c74: uart_event_task(void*) at /home/tommaso/esp/workspaces/tinyproto_ws/tinyproto_52/main/main.cpp:55
0x400871bd: vPortTaskWrapper at /home/tommaso/esp/v5.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

E (26895) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (26895) task_wdt:  - IDLE0 (CPU 0)
E (26895) task_wdt: Tasks currently running:
E (26895) task_wdt: CPU 0: uart event task
E (26895) task_wdt: CPU 1: IDLE1
E (26895) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400DB32F:0x3FFB0E40 0x400DB6F4:0x3FFB0E60 0x400830B9:0x3FFB0E90 0x400D7091:0x3FFB8420 0x400D71E5:0x3FFB8440 0x400D6865:0x3FFB8460 0x400D5E65:0x3FFB8490 0x400D5C74:0x3FFB84B0 0x400871BD:0x3FFB84E0
0x400db32f: task_wdt_timeout_handling at /home/tommaso/esp/v5.2/esp-idf/components/esp_system/task_wdt/task_wdt.c:441
0x400db6f4: task_wdt_isr at /home/tommaso/esp/v5.2/esp-idf/components/esp_system/task_wdt/task_wdt.c:515
0x400830b9: _xt_lowint1 at /home/tommaso/esp/v5.2/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x400d7091: hdlc_ll_read_end at /home/tommaso/esp/workspaces/tinyproto_ws/tinyproto_52/components/tinyproto/src/proto/hdlc/low_level/hdlc.c:447
0x400d71e5: hdlc_ll_run_rx at /home/tommaso/esp/workspaces/tinyproto_ws/tinyproto_52/components/tinyproto/src/proto/hdlc/low_level/hdlc.c:516
0x400d6865: tiny_fd_on_rx_data at /home/tommaso/esp/workspaces/tinyproto_ws/tinyproto_52/components/tinyproto/src/proto/fd/tiny_fd.c:841
0x400d5e65: tinyproto::IFd::run_rx(void const*, int) at /home/tommaso/esp/workspaces/tinyproto_ws/tinyproto_52/components/tinyproto/src/TinyProtocolFd.cpp:97
0x400d5c74: uart_event_task(void*) at /home/tommaso/esp/workspaces/tinyproto_ws/tinyproto_52/main/main.cpp:55
0x400871bd: vPortTaskWrapper at /home/tommaso/esp/v5.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

Even the generator example has this kind of issue, I haven't tested the loopback example though

lexus2k, 2024-02-18:
In the library there are no blocking calls at `hdlc_ll_run_rx at /home/tommaso/esp/workspaces/tinyproto_ws/tinyproto_52/components/tinyproto/src/proto/hdlc/low_level/hdlc.c:516`

The only place to lock up is inside `tiny_fd_on_rx_data`, which calls `hdlc_ll_run_rx` in the loop. It will exit only if all bytes are processed. It may lock if negative number of bytes is passed. 

Could you please try the code from `cpp_api` branch? Does it give the same result?

Tizio1234, 2024-02-22:
I'm getting the same result even when using cpp_api branch

lexus2k, 2024-02-24:
@Tizio1234 

`-DTINY_HDLC_DEBUG=1`, `-DTINY_FD_DEBUG=1`, `-DTINY_DEBUG=1`, `-DTINY_LOG_LEVEL_DEFAULT=5` could help to find a root cause.
The FD communication requires both tx and rx threads to run continuously. In your case you use  `vTaskDelay(pdMS_TO_TICKS(30));` to make a task have a sleep. 

There are 2 modes of the library: Single thread mode and Multithread mode.
In single thread mode, the main loop / or main task should execute `run_rx()`, `run_tx()` sequentially in the loop.
In multithread mode, you need to have one rx task, and one tx task, calling functions in the loop.