lexus2k/tinyproto #44: Arduino Due - Serial3 not working

racarla96, 2024-03-13:
I am trying to see how the library works and I am trying to see what happens onReceive by trying to see the output through another serial port (arduino Due), this would be the code. Is there something I'm doing wrong or don't know? Thank you

```
#include <TinyProtocol.h>
// We need this hack for very small controllers.
#include <proto/fd/tiny_fd_int.h>

/* Creating protocol object is simple. Lets define 48 bytes as maximum. *
   size for the packet and use 4 packets in outgoing queue.             */
tinyproto::Fd<FD_MIN_BUF_SIZE(48, 4)>  proto;

void onReceive(void *udata, tinyproto::IPacket &pkt)
{
  Serial3.println("-----------------");
//  for (int i = 0; i < pkt.size(); i++)
//  {
//    Serial3.print(pkt.getByte());
//    Serial3.print(" ");
//  }
//  Serial3.println();
  
  if (proto.write(pkt) == TINY_ERR_TIMEOUT )
  {
    // Do what you need to do if there is no place to put new frame to.
    // But never use blocking operations inside callback
  }
}

void setup()
{
  /* No timeout, since we want non-blocking UART operations. */
  Serial.setTimeout(0);
  /* Initialize serial protocol for test purposes */
  Serial.begin(115200);
  /* Lets use 8-bit checksum, available on all platforms */
  proto.enableCheckSum();
  /* Lets process all incoming frames */
  proto.setReceiveCallback( onReceive );
  /* Redirect all protocol communication to Serial0 UART */
  proto.begin();

  Serial3.begin(115200);
  Serial3.println("START");
}

void loop()
{
  if (Serial.available())
  {
    Serial3.write(".");
    Serial3.flush();
    proto.run_rx([](void *p, void *b, int s)->int { return Serial.readBytes((uint8_t *)b, s); });
  }
  proto.run_tx([](void *p, const void *b, int s)->int { return Serial.write((const uint8_t *)b, s); });
}
```

racarla96, 2024-03-23:
The library seems to work to communicate with the example program with the stable v1 api, but it does not allow any further action to be executed, it seems to get stuck in the call:
-> proto.run_rx([](void *p, void *b, int s)->int { return Serial.readBytes((uint8_t *)b, s); });
And I don't know how to continue to debug the problem.

lexus2k, 2024-04-01:
@racarla96 Did you set timeout for Serial object to zero / or small value.

By default `Serial` has too big timeouts (maybe infinite one)

alex-eri, 2024-09-03:
I using hdlc

I set small timeouts. btw ReceiveCallback not fires on rx.

rx

```

  while (Serial1.available())
  {
    Serial.println();

    size_t len = Serial1.read(rx_buffer, sizeof(rx_buffer));

    for (size_t i = 0; i < len; i++)
    {
      Serial.print(rx_buffer[i],16);
      Serial.write('\t');
    }

    size_t rx_offset=0;
    size_t rx_len=len;

    Serial.println();
    int err;
    err = hdlc.run_rx(&rx_buffer+rx_offset, rx_len);

    rx_len-=err;
    rx_offset+=err;
    Serial.println(err);

    err = hdlc.run_rx((&rx_buffer)+rx_offset, rx_len);
    Serial.println(err);

  }
  
  
  ```
  
  Check sum not filled on TX
  
  ```
     hdlc.write((const char *)(&net_pkt), sizeof(packet_t));

    int len = 0;
    while (true)
    {
        len = hdlc.run_tx(tx_buffer, sizeof(tx_buffer));
        if (len == 0)
        {
            break;
        }

        for (int i = 0; i < len; i++)
        {
            Serial1.write(tx_buffer[i]);
        }

        for (int i = 0; i < len; i++)
        {
            Serial.print(tx_buffer[i], HEX);
            Serial.print('\t');
        }

        Serial.println();
    }
  
  ```